sonocotta/esparagus-media-center #7: Can I buy the pre-flashed device in Europe and in Euro currency?

Dricc123, 2024-05-11:
the title says it all. Thanks.

anabolyc, 2024-05-11:
Boards are not generally available while the crowd supply campaign is
running, but i do offer samples to beta testers. Please drop me a line at
***@***.***

Dricc123, 2024-05-11:
I will wait for now, but thanks. My question was more about whether the boards will be available with shipping within Europe (and sold in euros) one day? Your manufacturing takes place in Europe, correct?

anabolyc, 2024-05-11:
Yes, absolutely. As someone on as all backers orders are shipped, i’ll
reopen Tindie sales. I ship from Poland, where I assemble, flash ant test
every board myself :)

closed 2024-05-12 by anabolyc

Dricc123, 2024-05-13:
But Tindie shows the price in USD currency. Can I change that?

anabolyc, 2024-05-13:
Tindie only charge USD, but as I said you can contact me directly for custom order.

Dricc123, 2024-05-13:
I can't find your contact details on your website and Github hides the email address that you wrote down here above

anabolyc, 2024-05-13:
Oh, sorry, I didn't realize that (probably my email client did)! It is andriy@sonocotta.com. Also, there is a Discord link in my GitHub profile.