sonocotta/esparagus-media-center #8: Schematics show "xy ESP" versions with OLED Display ?

RobertK66, 2024-09-17:
The hardware section shows PDF schematics with Titles "HiFiEsp 32", "Loud ESP Micro", "Louder ESP". All of them do have a OLED which is not available on the hardware described in this repos!

I think there should be "Esparagus" versions available for that. I assume most of the content will be same (especially the JTAG connector ;-) ) but would like to get information on all available IOs   (LED?, IR?, relay driver?)...

PS: got my 2 devices last week and so far everything works perfect, thx.

anabolyc, 2024-09-17:
Hey @RobertK66 

Those names that you mentioned are prototype names that I used before the Esparagus name appeared. But you have a valid point, to my own surprise, I didn't publish the latest revision's hardware files. Sorry about that, I'm doing it now. Please have a look and let me know if something is missing.

anabolyc, 2024-09-17:
Regarding OLED option.

Every board has an OLED direct-solder connector on the bottom side of the board that is compatible with popular OLED panels (you may find them by the 30-pin connector). It is there, it is tested and working, but since it doesn't fit the case, I never advertised it. But feel free to give it a try, if you have an idea on how to fixture it.

[image]

RobertK66, 2024-09-17:
Thx. for your fast reply! seems to be right now (the prints also show the 'IR - nose' sticking out of the case now ;-.)!)

closed 2024-09-17 by RobertK66