sonocotta/esparagus-media-center #10: Louder ESParagus over POE++

potelux, 2024-09-27:
I was working on a PCB of my own that would house an ESP32 WROVER, contain a class-D amp for my speakers, and be powered over POE.  I then came across this project and was very excited to see the work you have done and how similar it was to my end goal.

Have you considered/would you be interested in allowing the Louder ESParagus to be powered over POE++ which supports between 50 and 70 watts (depending on the version of POE++)?  If so, are you welcoming help with the PCB design?

If you have not considered this and are not interested in doing so, would you consider selling the case without the PCB?

potelux, 2024-09-27:
In this same vain, I can see the use for a POE ESParagus Medialink with RCA or TOSLINK out.  I would be happy to contribute to either if you would have an interest as well.

anabolyc, 2024-09-27:
Hey @potelux 

Some time ago I decided against PoE since a typical PoE appliance would not deliver enough power. And 100 W+ over PoE is a bit exotic, at least currently. Another concern is the power line quality considering it goes through at least a few meters of cable alongside a high-frequency digital signal. 

But I don't mind giving it a try. Do you have a schematics that I can look up. Do you aim to design it yourself with my help or other way around?

Regarding RCA/TOSLink, afaik you simply need a digital pin connected to it, so it is only an issue of space on the back side to place it, probably can be solved one way or another. Again, if you can share some schematics, it will bring me up to speed.

I was planning to make a no-PD version anyway, so I might prepare alternative test boards as well

potelux, 2024-09-30:
> Hey @potelux
>
> Some time ago I decided against PoE since a typical PoE appliance would not deliver enough power. And 100 W+ over PoE is a bit exotic, at least currently. Another concern is the power [...]
You have a good point here.  If that much power is needed, an external supply may be worth it.  To provide some clarity, I am specifically thinking about the application of rack deployment for whole-home audio.  The devices would be running Snapcast and controlled via MusicAssistant.  Using POE provides allows for clean cable management, but would require a very expensive switch to provide enough power.  

> Regarding RCA/TOSLink, afaik you simply need a digital pin connected to it, so it is only an issue of space on the back side to place it, probably can be solved one way or another. Again, if you can share some schematics, it will bring me up to speed.
It could still be beneficial to do a POE ESParagus MediaLink with a more typical HiFi output like TOSLink(or XLR) in a rack setup since the power draw is so minimal and the cable management of several power supplies and cables would be a bit chaotic.  In other words, a simple version of the MediaLink with only two ports, namely, an RJ45 and a TOSLink(or XLR).  Connect a couple and rack mount them, then connect to an external amplifier and voila!

> But I don't mind giving it a try. Do you have a schematics that I can look up. Do you aim to design it yourself with my help or other way around?
This is something I am currently designing myself.  I don't have any schematics worth sharing yet as I am still in the prototyping phase (but I would be willing to share some reference schematics I am using for POE).  My intent on informing you is to see if there is an obvious reason I am overlooking that precluded you from pursuing this.  Additionally, if I end up using your work as a reference for my own, I would like to provide some value in return.

potelux, 2024-09-30:
This is a different product all together, but I figure you may benefit from the idea.  I also intend to build the inverse of your MediaLink.  A POE device that inputs audio and acts as a source for a SnapCast server.  The outcome is a modular set of audio inputs and outputs that is software configurable.

zendes, 2024-09-30:
> It could still be beneficial to do a POE ESParagus MediaLink with a more typical HiFi output like RCA (or XLR) ...

To me this would actually be the less useful one of them all. If you're going to run a cable to an amp, you are most likely already near a power source.

The one where I can see more value (assuming you can deal with the power quality issues that were mentioned before) is a remote installation where you only run a POE ethernet cable, and then locally connect a louder to a set of ceiling speakers and don't have to worry about also supplying power to the device.

anabolyc, 2024-10-01:
> You have a good point here. If that much power is needed, an external supply may be worth it. To provide some clarity, I am specifically thinking about the application of rack deployment for [...]

For that matter, I'm currently working on the 8 stereo channel DAC setup with Pi5, you'd need to wire speaker cables only, having everything "smart" in one place on the rack.

Your approach have a place too for sure, but I still think running power over ethernet cables is not a perfect approach for audio applications 

> It could still be beneficial to do a POE ESParagus MediaLink with a more typical HiFi output like RCA (or XLR) in a rack setup since the power draw is so minimal and the cable management of several power supplies and cables would be a bit chaotic. In other words, a simple version of the MediaLink with only two ports, namely, an RJ45 and a RCA (or XLR). Connect a couple and rack mount them, then connect to an external amplifier and voila!

For digital audio that should work quite well. No need for lots of power, no need for clean voltage either.  How do you plan to decode it to the speaker?

> This is something I am currently designing myself. I don't have any schematics worth sharing yet as I am still in the prototyping phase (but I would be willing to share some reference schematics I [...]

If there is anything I can help you with, let me know. Obviously, I'd be happy to know that my schematics or documentation helped you in one way or another.

potelux, 2024-10-01:
> > It could still be beneficial to do a POE ESParagus MediaLink with a more typical HiFi output like RCA (or XLR) ...
> 
> To me this would actually be the less useful one of them all. If you're going to run a cable to an amp, you are most likely already near a power source.

While I think that stands true in most applications, I have experience in large AV rack installation and power cable management is often a greater challenge.  Even though the power strips are in close proximity, the number of outlets is limited.  The cost of additional filtered/clean power rack strips is higher than the cost of upgrading a switch to a POE switch.  Since all the devices would already by powered by a CAT6 cable, using that for power simplifies the set up.  

I know this is a more niche case, and I assumed that was why you hadn't already created such a product.  My use case is not the consumer who will purchase one or two units, but rather the installer who uses 10-20 in a rack.

> The one where I can see more value (assuming you can deal with the power quality issues that were mentioned before) is a remote installation where you only run a POE ethernet cable, and then locally [...]

This is also a great option; one I hadn't really considered.  If this is something that could be valuable to you (maybe even a future product), then I'll start exploring this avenue.

anabolyc, 2024-10-02:
Cool ideas right there. To reiterate, if there is anything I can help you with, let me know. I do not have too much free time these days, but always happy to share as much as I can.

potelux, 2024-10-02:
> Cool ideas right there. To reiterate, if there is anything I can help you with, let me know. I do not have too much free time these days, but always happy to share as much as I can.

Sounds good!  I am planning to build off of your work, so I was just reaching out to see where your head was at.  I want to be sure I am not wasting time and that, if possible, I could pursue something that benefits you as well.  

Thanks for your time so far!