sonocotta/esparagus-media-center #11: Missing instructions to flash Squeezelite itself

zendes, 2024-09-27:
Hi,

I received my loud & louder order, and I'm trying to flash Squeezelite onto them using https://sonocotta.github.io/esparagus-media-center/. Your flashing instructions and website only get to the squeezelite recovery mode, but don't contain instructions on how to get the board config on there or which firmware update url to use. Are there instructions to continue from https://github.com/sonocotta/esparagus-media-center?tab=readme-ov-file#how-to-flash-and-configure 

It appears there are extra steps between "Exit Recovery" and "You can use it now". The loud & louder devices don't show up in the "Known Boards" section. Is there a special firmware URL I should use?

Thanks,
Dennis

anabolyc, 2024-09-27:
Hey @zendes 

The flasher tool has all the board-specific config baked into it, so you don't need to change anything in the NVS settings unless you have some specific task in mind. 

If you are stuck on a specific step, please attach a screenshot.

zendes, 2024-09-27:
Here are the steps I'm going through:

1. I go to https://sonocotta.github.io/esparagus-media-center/
2. I erase and install Louder Esparagus on the device
3. I connect to louder-esparagus wifi with squeezelite pw, and connect to my wifi
4. I reconnect to my wifi, and go to the device IP
5. This one I'm not sure about. Should I now click "Exit Recovery" or flash something in Updates?

If I click Exit Recovery the device ends up in a boot loop, and I don't know what firmware image to flash if I need to update it.

I've attached the console logs from the device from flash to exit recovery and the boot loop:
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/17169488/esp-web-tools-logs.txt)

anabolyc, 2024-09-27:
You're doing everything right. You should simply exit the recovery mode. Boot loop is not expected here. Afaik it would normally reboot if unable to connect to wifi or ethernet. But once it is connected it should stay on for a longer time.

This could be connected to [that bug](https://github.com/sle118/squeezelite-esp32/issues/414). Can you please try to clear led_vu_config nvs setting and see if it helps

zendes, 2024-09-28:
Thanks. I’ll go find that a try and I’ll report back hopefully by end of the weekend.

zendes, 2024-09-29:
I can confirm that clearing the led_vu_config NVS setting works, and the boot loop is no longer there. It also appears to be fixed in 1681, so flashing that right after connecting to the device wifi without clearing the NVS led_vu_config also works

anabolyc, 2024-09-29:
Thanks for letting me know. I've updated the readme to keep it visible. Closing it now

closed 2024-09-29 by anabolyc