sonocotta/esparagus-media-center #13: Unable to load Firmware on linux using Google Chrome

rjohnsonbade, 2024-12-10:
Hello, I have a Louder Esparagus and I am trying to load the firmware on to the device.

I am connected via USB-C and am using Chrome, when I "Connect to device" I get the browser dialog to select the serial port, "ttyUSB0" is available and I select it.

The following alert is displayed by the browser -

[image]

If I try again I can see that the serial option says "Paired" -

[image]

I am running Fedora Workstation 41 on a Framework 13 laptop.

Any ideas on how to diagnose the issue?
I am comfortable running commands in the terminal, however I don't have much experience communicating over serial ports.

anabolyc, 2024-12-10:
Hey.

Please check the ownership of the /dev/ttyUSB0 device and your own groups. Do you have the rights to speak to device yourself? You can try it simply running `screen /dev/ttyUSB0 115200`

Other thing, please do the same with Chrome Dev Tools open, it should spit out little more details in the console

rjohnsonbade, 2024-12-11:
Thanks for the quick reply.

I checked ownership and I could see it was `root dialout`, I added my user to the `dialout` group, sadly no change.

The only logs I'm getting in the Chrome Console are related to some other devices (kb + mouse) -
```
Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001812-0000-1000-8000-00805f9b34fb, name='ERGO K860'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000180f-0000-1000-8000-00805f9b34fb, name='ERGO K860'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001801-0000-1000-8000-00805f9b34fb, name='ERGO K860'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000180a-0000-1000-8000-00805f9b34fb, name='ERGO K860'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001800-0000-1000-8000-00805f9b34fb, name='ERGO K860'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001812-0000-1000-8000-00805f9b34fb, name='MX Vertical'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001801-0000-1000-8000-00805f9b34fb, name='MX Vertical'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000180f-0000-1000-8000-00805f9b34fb, name='MX Vertical'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001800-0000-1000-8000-00805f9b34fb, name='MX Vertical'
esparagus-media-center/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000180a-0000-1000-8000-00805f9b34fb, name='MX Vertical'
```

Trying to attach via screen just does the following -
```
russjb@fedora:/dev$ screen /dev/ttyUSB0 115200
[screen is terminating]
```

--

Ok so I chowned `/dev/ttyUSB0` to my user, and then tried to connect via screen, et voila! It connected! I then tried via Chrome and it didn't work, but I realised my screen session was still there and occupying the port. Took me a sec refreshing my screen knowledge but after doing that I was away!

[image]

Thank you!

All I needed was to chown the device to my user; I guess even with my user added to the dialout group it wasn't enough? I wonder whether that's something to do with how chrome is launched?

rjohnsonbade, 2024-12-11:
Thoughts on adding something like this?

[image]

anabolyc, 2024-12-12:
@rjohnsonbade 

Assuming you did reboot and it didn't help, please have a look at this [article](https://docs.platformio.org/en/stable/core/installation/udev-rules.html). It should get you permanently fixed.

> Thoughts on adding something like this?

Sure, makes sense. Can you please check the solution above, and I'll update the hint

rjohnsonbade, 2024-12-16:
Hi @anabolyc , so I just plugged my Louder Esparagus in to test the above.
I can see it's owned by root:dialout, and confirmed my user is (still) in the dialout group.

Now when I plug it in it Connects successfully right away, I connected to the console and it is definitely communicating without any problems.

I did not need to specify any udev rules like in the article.

I just removed myself from dialout and restart, and again received the error above.
Adding myself to dialout group yet again and restarting solved the problem.

So my issue was simply device ownership, you either need to add yourself to dialout group and reboot; or you need to change the ownership of of the device to your user, which does not require reboot, but is probably less correct.

closed 2024-12-18 by anabolyc