sonocotta/esparagus-media-center #16: Only sound from left channel

iliessens, 2024-12-21:
Hi,

I've recently noticed that my Esparagus HiFi Medialink (the official revision E board from Tindie), only outputs audio to the left channel. The right channel audio is not heard at all (no downmix either).

Sadly, I can't remember if this is new or has always been the case.

This happens both through Bluetooth and LMS, with different music sources and output devices. So it does seem to be related to the board or its firmware. (Running Squeezelite I2S-4MFlash-1681.)
The DAC pin assigments in the GUI seem correct with the schematic. (CLK= 26, WS = 25, DATA = 22)

A quick check of the PCB doesn't reveal any missing components or bad solder joints.

Any ideas what could be causing this? Apart from this it works very well.

anabolyc, 2024-12-21:
The most common cause is a bad connection around 3.5 mn jack, either bad
solder joint, or bad connection inside. Can you try to wiggle it a little
and check if there is anything audible on the speaker

Second common reason are bad headphones:) in case you use them, please
check with any other device to make sure.

Do you have equipment to check signals on the schematics for the next steps?

iliessens, 2024-12-21:
I tried to wiggle the connector and tap the solder joints a bit, but nothing changed.
But when I broke out my oscilloscope I noticed the signal was present until C14, but not on the connector. And indeed, a quick blob of solder on the pad of the 3.5 mm jack fixed it!

Thanks for your swift and helpful reply! The audio quality is really good for an open source device as well 😊

closed 2024-12-21 by iliessens