sonocotta/esparagus-media-center #18: Medialink, no output with squeezelite-esp32 on Spotify

MrNonoss, 2025-01-12:
Hi there,

I am facing an issue with the MediaLink not working reliably with the squeezelite-esp32 firmware.

I received it pre-flashed and it worked out of the box after configuring wifi.

I am using it to output sound via the jack to an external amp, with Spotify as a source through cspot. Not Bluetooth that I disabled.

But I noticed that after a certain amount of days, even though the Esparagus is shown in the available speakers in the Spotify app, it does not output any sound.

Disconnect and reconnect the MediaLink was a fix. 
But I decided to update the firmware.

I went to the web interface, spotted that the pre installed firmware was I2S-4MFlash-xxx (I don't recall the original build number), so I updated to the latest build at the time of writing I2S-4MFlash-1681.

Now, it did not fix the issue and even worse, sometimes the Esparagus is even not showing in the Spotify app.

Any idea on how I can troubleshoot this or if it can be a hardware issue?

Many thanks for the help.

anabolyc, 2025-01-12:
Hey @MrNonoss 

The best place to start is to connect to the device via USB and start capturing Serial logs, they might show some relevant errors. The most common issue with Spotify is some API calls to Spotify failing, sometimes it is caused by Spotify implementing braking changes on their side (I witnessed this at least once). 

I would very doubt it is a hardware issue since PCM5100 DAC is a very simple device, doesn't require any configuration, and cannot be disabled to shut down. Generally, It is very reliable.

MrNonoss, 2025-01-13:
Thank you for your answer.

I collected serial logs: [esp-web-tools-logs.txt](https://github.com/user-attachments/files/18398429/esp-web-tools-logs.txt)
I can see some errors like this one:

```
E (157774) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (157774) task_wdt:  - IDLE (CPU 1)
E (157774) task_wdt: Tasks currently running:
E (157774) task_wdt: CPU 0: IDLE
E (157774) task_wdt: CPU 1: cspot_player
E (157774) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401E2C56:0x3FFBED40 0x40084889:0x3FFBED60 0x40270227:0x3FFBD5B0 0x400DD261:0x3FFBD5D0 0x4027E6CB:0x3FFBD5F0 

E (157774) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008F5DD:0x3FFBF570 0x40084889:0x3FFBF590 0x4026CE77:0x3FB01AD0 0x400F7A45:0x3FB01AF0 0x400F8258:0x3FB01B10 0x400F8335:0x3FB01B40 0x40109871:0x3FB01B60 0x401098A5:0x3FB01BD0 0x4015CD00:0x3FB01C00 0x401088AA:0x3FB01CB0 
```

Unfortunately, I do not understand it.

anabolyc, 2025-01-13:
Looks like the cspot process got stuck. Can you please report it to the squeezelite developers [here](https://github.com/sle118/squeezelite-esp32/issues)

MrNonoss, 2025-01-15:
Many thanks, sure I will and I will update here.
Can you confirm we should flash the 32 bits version and not the 16?

2.1681-32 | 9/12/24, 11:27 AM | I2S-4MFlash | master-v4.3 | 32
2.1681-16 | 9/12/24, 11:27 AM | I2S-4MFlash | master-v4.3 | 16

MrNonoss, 2025-01-15:
Copy that. I will re-flash then.
Many thanks again.