sonocotta/esparagus-media-center #21: Using bluetooth as a master device?

Wildcarde, 2025-03-18:
I'd like to put a small esparagus device in my garage to drive my speakers in there over the standard stereo out + ir blaster. However while using equipment I am wearing bluetooth capable headphones, I'd like to be able to pair these with the esparagus device directly to mirror whatever is playing over the speakers rather than just cutting off whatever I'm listening to.  I'm not sure if this is even possible with the equipment available to the device, any thoughts on how this would work?

anabolyc, 2025-03-18:
Hey @Wildcarde. I think you're asking for a bit too much from the humble ESP32. First, It can only work as a BT player, not the source device AFAIK. I'd try looking into Raspberry direction for that kind of custom setup, although classic BT is a point-to-point type of communication, so mirroring probably is not something that will work out-of-the-box

Wildcarde, 2025-03-18:
fair enough, I can probably make it work with pipewire on a 'proper' linux setup, I was just hoping to have something lighter weight to target but we work with what we've got!

closed 2025-03-18 by Wildcarde

Wildcarde, 2025-03-18:
second followup (sorry), have you tried getting this working on system? https://github.com/espressif/esp-adf It looks like a full audio dev environment and does include both source and sink modes for bluetooth.

anabolyc, 2025-03-18:
Yes, I've tried, and it works well. But I'm not up to implement anything beyond provided examples. In fact I'm working on the adf driver implementation for TAS5805M, but I think base implementation is there already.