sonocotta/esparagus-media-center #23: A question about esphome custom components

jeremiahsb, 2025-03-20:
With the esphome releases in 2025, the team has removed custom components:
https://esphome.io/guides/contributing#a-note-about-custom-components

Are there any plans to migrate the esphome yaml to use external components?
https://esphome.io/components/external_components

anabolyc, 2025-03-20:
Hey @jeremiahsb 
Absolutely, yes. It is a work in progress and tracked in the sister repo [here](https://github.com/sonocotta/esp32-audio-dock/issues/35). The working branch is [here](https://github.com/sonocotta/esp32-tas5805m-dac/pull/1)

I'm trying to add DSP capabilities at the same time, like EQ so it takes a bit more time than I'd hope for.

anabolyc, 2025-05-23:
While work is still in progress, I'm posting the config that works for me (external component work is not finalized yet). It is based on esp-idf and compared to the Arduino implementation, it works much better, no glitches and much more stable, at least based on my limited experience. bonus points, this implements a 15-band equalizer, with -15..+15dB regulation, works really amazing, really looking forward to having it finalized

```yaml

substitutions:
  name: esphome-web-dc73f8
  friendly_name: louder-esp32
  task_stack_in_psram: "false"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.11.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  # on_boot:
  #   priority: 600  # After most components are initialized
  #   then:
  #     - lambda: |-
  #         id(tas5805m_dac).set_volume(int(id(current_volume)));
  #         ESP_LOGI("tas5805m", "Restored volume on boot: %.0f", id(current_volume));
esp32:
  board: mhetesp32minikit
  flash_size: 8MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"

      CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC: "y"
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "y"  # TLS1.3 support isn't enabled by default in IDF 5.1.5

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
  platform: esphome
  password: !secret esphome_ota_password

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  ap:
    ssid: "$name Hotspot"
    password: !secret esphome_ap_password

captive_portal:

psram:
  speed: 80MHz

#  - source: /config/esphome/esp32-tas5805m-dac/esphome/components
#  - source: /config/esphome/external/esphome-components-test/components
external_components:
  - source: github://mrtoy-me/esphome-components-test@dev
    components: [ tas5805m ]
    refresh: 0s

# script:
#   - id: set_volume
#     parameters:
#       level: float
[... 392 lines cut ...]
              - media_player.is_playing:
              - lambda: return (!id(eq_gain_refreshed));
          then:
            - lambda: |-
                id(tas5805m_dac).refresh_eq_gains();
                id(eq_gain_refreshed) = true;

    turn_off_action:
      - lambda: id(tas5805m_dac).set_eq_off();

interval:
  - interval: 30s
    then:
      - if:
          condition:
            - switch.is_on: enable_louder
            - for:
                time: 120s
                condition:
                  or:
                    - media_player.is_idle: louder_mediaplayer
                    - media_player.is_paused: louder_mediaplayer
          then:
            - switch.turn_off: enable_louder
```