sonocotta/esparagus-media-center #25: Esperagus Loud esphome yaml

txwindsurfer, 2025-04-30:
I either have a defective esperagus loud or the .yaml file in firmware is not correct.  I have 2 issues:

1) If I connect Speaker R, the output is only a very loud static when a message command is issued.  If I only connect Speaker L then there is no problem and the message plays correctly.  I have tried 2 different cables therefore it does not appear to be the cables. 

2) The logic within loud-esperagus.yaml that is intended to use the speaker switch on GPIO13 does not function correctly because the message is repeated over and over until a media player command to pause or stop is issued.  Also there are no log entries being generated.  The restart button works correctly.

anabolyc, 2025-04-30:
Hey @txwindsurfer 

Let's first check if there is a hardware issue with the DAC. Do you have the possibility to test it with squeezelite-esp32? Latest ESPHome updates came with a bunch of weird issues, so I'd like to test it outside it first.

There was a question regarding the Enbale pin, but looks like it is resolved already

txwindsurfer, 2025-05-01:
I did hook up the Loud as shipped and I think it worked fine with squeezelite.  I do think the ESPHome Arduino updates could be the problem.  Do you have a earlier version that you suggest I try?

anabolyc, 2025-05-02:
To be honest, I'm not sure what the latest version I tested was. Can you please check the version you have, and I'll try to test with the same version and see if I can reproduce the issue. 

By the way, do you only have the glitches with TTS or with audio playback as well? Do you use Google TTS or Piper (I found latest work generally better)

anabolyc, 2025-05-03:
Hey @txwindsurfer I've updated the yaml, tested with both music assistant and google TTS. Note, I'm using ESPHome 2025.4.1

txwindsurfer, 2025-05-10:
Hi @anabolyc,  and thanks so much for all the help.  Sorry for the delay--traveling this week.  I upgraded to 2024.4.1 and I added the min_ version line and took out the octal line in the yaml.  Unfortunately, R+ still produces loud static.  I am willing to buy a 2nd Loud to see if it is a hardware issue because this is a very nice device for playing announcements from an Atom Echo over my home speakers instead of the tiny speaker on the Atom Echo.   Should I order a second or do you think there is more to try?

I would also add that using only the L side is ok for announcements; however, when I program the Atom Echo to announce directly to the Loud media player, the response is very quick but in the high pitch, high speed Chipmunk sound (mismatched audio rate).  However, if I program the Atom Echo to announce to the Music Assistant Loud media player, the announcement works but only after  a 3-5 second delay.  I have tried different TTS engines like cloud_say, piper, nabu, etc.  I can live with the delay if needed, but wondered if you had any thoughts on this.

anabolyc, 2025-05-11:
Hey @txwindsurfer 

- Before you order another unit just to confirm if the hardware is okay, I'd advise testing it with another software (like squeezelite) and see if the problem persists. If possible, can you attach a photo of the wiring you've made and a video recording of the sound that is defective?

- I think delay is caused by recent update, which introduced HA to re-code all TTS announcements to standard data format and rate. Before that update some TTs engines would simply fail to produce audio, now they work much more stable but with a few seconds delay. At least that is my observation.

txwindsurfer, 2025-05-16:
I flashed Loud Esperagus Squeezelite from https://sonocotta.github.io/esparagus-media-center/   Hooked up everything and could login to the console and see the logs.  No errors that I could determine.  However while Spotify and Music Assistant could find Loud Esperagus no sound was coming from the speakers.  Reflashed ESPHome and the Espergus Loud yaml.  Sound was coming out of the speakers in Music Assistant but if I attached wires to the either ground there a slight increase in volume and some distortion which appeared to be overdriving the powered speakers.  Just hooking up L+ and R+ worked ok with the powered speakers.  When attaching to unpowered speakers (no amp to process) I had to hook up only L+ and L-.  Any wire on R+ or R- produced very loud static.  Too many variables to diagnose my issues (esp32, yaml, esphome, wires, powered speakers vs unpowered speakers, etc.)  Anyway just closing the issue as it works for my use case: announcing Atom Echo Voice Assist command output to my installed home speakers.

closed 2025-05-16 by txwindsurfer

anabolyc, 2025-05-17:
If I get it right, you tried to make it work with active (powered) speakers, which would definitely cause distortions. Lous Esparagus has a power driver at the output and would drive it 10V peak-to-peak, way too much for active speakers. For that use case, you'd be able to use HiFi Esparagus, which has line-level output and **require** powered speakers. Loud should be connected to the drivers directly. Anyway, I'm glad it worked out the way you see it fit.